sinara-hw/DIOT_EEM_Adapter #1: "Smart" backplane for EEM/Kasli ecosystem

dhslichter, 2019-01-03:
Moving my comment from #204 to a new issue:

Might it be better in the long haul to have Kasli's functionality actually just integrated on the backplane? You would have a slim "dumb" card that brings SFP and coax connections to the front panel, and connects them to the backplane via some high-performance but lower-pin-count connector, suitable for gigabit signals. Then the Artix FPGA, clock distribution, etc all live on the backplane, and you don't need crazy fat connectors just to route all the EEM signals and clocks off the Kasli and on to the backplane. This would also free more front panel space, because if you are adding 96-DIN mezzanines to Kasli then this obviously will make it wider, which may be undesirable for some.  

The only downside I can see is that if your FPGA on the backplane goes bad for some reason, it's more work to change it out....but this seems like it should not be a very common occurrence.

gkasprow, 2019-01-10:
I've ordered [this chassis](https://schroff.nvent.com/en/schroff/24579-416). I will make sure the connectors are symmetrical and produce the adapter board.  I will also switch card depth from 160 to 220mm. Guys @CERN claim that it is possible.

jordens, 2019-01-11:
Nice. The chassis core is just the plain old standard, the front and back rails are with the longer cPCI lips. And the middle rails are movable every 60mm. Get the longer plastic rails, they seem to be available in 220 mm with cPCI coding. I guess you won't have much space in the back anymore which could be a problem for the power connector and the 160 mm power supply might need some hackery to connect the backplane.

gkasprow, 2019-01-11:
I have 80mm on the back - there is space for RTM option.
Supply has dedicated backplane, so we can mount two sets of rails and leave existing plastic rails.
In this way, supply backplane and plastic rails would reside on 160mm rails while rest would be attached to additional rails installed on on 220mm depth.

gkasprow, 2019-01-11:
Anyway, I verify it with real HW. Delivery will be within 4 weeks.

gkasprow, 2019-01-11:
When I look at the crate rear side, the holes seem to be not placed evenly every 60mm
[image]
Anyway, who says we have to use 220mm length?:)

gkasprow, 2019-01-17:
Some ideas came to my mind. We plan to develop together with the CERN CPCIS controller based on Kasli, but with additional FMC interface and ZynQ. With plugged quad SFP FMC we will get essential Kasli functionality. But we will have to switch to the FFG900 package which means we will have more transceivers (16). 8 GTX will be routed to FMC to enable 8x SFP FMC that already exists. The other 8 GTX can be routed to the backplane. We can reserve some CPCIS slots as DRTIO-capable. And such slots can be equipped with DRTIO-connected modules as well. 
We will also develop 84HP backplane with 8HP slots. We can install DRTIO-capable slots between 8HP slots. In this way, one could combine 8HP modules like Sampler with advanced 4HP DRTIO and standard 4HP EEM modules. Up to 16 various modules could be installed.
Just an idea to discuss.

gkasprow, 2019-01-20:
@jordens @dtcallcock what do you think about such idea?
I specified it [here](https://github.com/sinara-hw/meta/wiki/Compact-PCI-Serial-EEM-ecosystem)

gkasprow, 2019-03-26:
Update:
- I agreed with CERN common specification of the open-source backplane. It would be a simple design with necessary signals to use with EEM cards. Boards can be 4HP (with spacers) or 6HP. 2 standard power supplies are supported. CERN will design the radiation hard power supply.
- the backplane can be installed at 160 or 220mm depth
- my CPCIS crate is on a way, I should get it in a few days.
- I'm thinking about a slightly different adapter design that would enable backplane while keeping 160mm card length. The idea is to make 3 adapters:
1. standard with EEM signals
2. active with PCIe or USB bridge so the whole ecosystem of modules could be used natively in CPCIS crate with standard x86 CPU
3. as above but with SpaceWire bridge.
4. simple passive adapter compatible with 220mm assembly

gkasprow, 2019-05-27:
After a long discussion, we agreed with CERN the specification for the backplane that fulfills all needs.
The contract for backplane design is almost in place.
The board will be designed using Kicad. It will be a simple, low-cost board that fits low-cost chassis but is fully compatible with CPCIS. The slot spacing is 6HP. If there is enough interest, we can make also 4 HP or 8 HP variant since the schematics will be the same.

gkasprow, 2019-07-14:
more or less final component placement. The adapter is mounted on top of Urukul
[image]
[image]

gkasprow, 2020-06-16:
Some photos of real HW. The front panel is 3D printed for the moment.
One detail needs fixing - the slot between upper and lower shield.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

gkasprow, 2020-06-16:
Entire shield/cover is low-cost laser cut. Just a few EUR for set.

hartytp, 2020-06-17:
Looks lovely! What's the cost of retrofitting that to an EEM? How does that compare to redoing the EEMs as native CPICs (not suggesting we should do that right now, just curious)?

hartytp, 2020-06-17:
oops, never mind, just saw your previous post. Is all that really just a few EUR? That seems very cheap.

gkasprow, 2020-06-17:
If we do it as native CPCIS, we would get rid of 4 connectors and 2 standoffs. NRE cost would be also slightly reduced because it would be only one board instead of two. But if we produce the adapters and shields in high quantities, the difference would be negligible. The most expensive are the CPCIS connectors anyway (10 + 9 EUR). I will improve the shield design to eliminate the nasty looking slots and reduce the stress on mezzanine connectors during extraction. The entire stress during insertion is already redirected by the shield.

hartytp, 2020-06-17:
Cool! It would be awesome to see an ARTIQ crate up and running with CPICS.

gkasprow, 2020-06-17:
I'm back to business so expect it soon.

gkasprow, 2020-06-20:
We finished the design of the completely open-source [CPCIS backplane](https://ohwr.org/project/diot-crate-hw/wikis/home) and chassis. It was made with Kicad.  It supports RTM option as well.

gkasprow, 2020-06-20:
The backplane can easily be tailored to specific use-cases. The same with chassis.

marmeladapk, 2022-12-19:
Closing - no action needed.

closed 2022-12-19 by marmeladapk