sinara-hw/DIOT_EEM_Adapter #3: add TVS to EEM I2C

gkasprow, 2019-11-23:


Tobs-bfi, 2020-05-05:
Hi, I've been following this project for a few days now, I saw that you have created a  cPCIs_adapter 
 for your EEM System. 
Do you have a design template for  CPCI Serial boards for designing a system slot and peripheral slot ?
I am looking for some information about CompactPCI-Serial cards? specifically card size and main P1 & P6  connector location.
The standard document is very expensive, and I do not need all of it; just mechanical data and signal names. Any help? 
Or do you have  any tutorial or short form specification?

Best regards Tobi

gkasprow, 2020-05-10:
please look at CERN DI/OT project on OHWR.  I used it as a template for my adapter. I paid 750$ for CPCI-S spec. It's a fair price. Is that what you call "very expensive" ?:)
https://www.picmg.org/product/compactpci-serial-specification-pdf-version/

Tobs-bfi, 2020-05-18:
Hi Greg,
oh thank you for your answer, no 750$ is not so expensive, your are right.
Its more expensive to create a special hardware or complete measuerment system.
Best Regrads

Tobias Lorenz

[cid:image001.png@01D6271F.747655B0]

BFI Betriebstechnik GmbH
Sohnstr. 65, 40237 Düsseldorf
T: +49 211 6707 272
F: +49 211 6707 297
E: tobias.lorenz@bfi-bt.de<mailto:tobias.lorenz@bfi-bt.de>

Amtsgericht Düsseldorf HRB 4055, Geschäftsführer: Dipl.-Ing. Christian Warias
Abschlusserklärung: Über das Internet versandte e-mails können unter fremden Namen erstellt oder manipuliert werden. Deshalb ist diese als e-mail verschickte Nachricht keine rechtsverbindliche Willenserklärung.
Disclaimer: Messages sent by e-mail can be manipulated by third parties. For this reason this e-mail is not legally binding.

https://www.picmg.org/product/compactpci-serial-specification-pdf-version/

marmeladapk, 2022-12-19:
@mkiepiela bump

mkiepiela, 2023-02-22:
TVS added.

marmeladapk, 2023-02-27:
I can't find these TVS.

mkiepiela, 2023-02-27:
Thought it was about TVS at DIOT connector.
Now TVS at EEM connectors also added.

closed 2023-02-27 by mkiepiela