sinara-hw/DIOT_EEM_Adapter #7: Power sequencing

kaolpr, 2021-12-16:
Could we implement in CPCIs version of Sinara some power sequencing?

The main advantage would be possibility of enabling peripherals one by one, what should help avoiding sudden current surge. Also peripherals could be identified before turning power on, what in some corner cases could prevent problems, in particular when used with DIOT Controller.

To be consistent with DIOT, we should preserve `SERVMOD` signal as peripheral I2C enabler, i.e. signal that is high only when controller is talking to the particular slot. Then, we could use one of the free GPIO expander pins to act as P12V enable signal.

This would obviously make boot-up procedure more complicated, so to avoid potential frustrations, we can add a dip switch that would force enable P12V.

gkasprow, 2021-12-16:
It makes sense. But do not implement hot-swap - we are using DC coupled signals.
So during the boot-up procedure, we must make sure that all the FPGA output signals are Hi-Z or driven low so the peripherals are not supplied from the FPGA by the signal lines. The LVDS drivers must be enabled after the peripheral module gains power.
If we don't do it properly, the peripheral drivers may get damaged. If we connect i.e. Phaser with FPGA, and bias the LVDS before powering on the FPGA supply, we may cause latchup.

marmeladapk, 2021-12-16:
> Also peripherals could be identified before turning power on, what in some corner cases could prevent problems, in particular when used with DIOT Controller.

That won't work, at least with FMC Carrier. It doesn't use standby power, P3V3_REG is generated from P12V0. That was by design.

gkasprow, 2021-12-16:
@marmeladapk you mean the CERN FMC Carrier? Maybe we should fix that. What's the purpose of the 5V standby power rail?

marmeladapk, 2021-12-16:
> What's the purpose of the 5V standby power rail?

Not sure.

> Maybe we should fix that.

I remember that Greg Daniluk specifically said that P3V3_REG is to be generated from P12V0.

gkasprow, 2021-12-16:
this we can fix by Sinara assembly variant

gkasprow, 2021-12-16:
In some Sinara boards that need to work in stand-alone mode, the P3V3MP is additionally generated from P12V0 as well.

kaolpr, 2021-12-16:
I talked to Greg today and they didn't anticipate any power sequencing. It was all about presence detection.

If support for sequencing is kept on the peripheral, we can implement it only in Sinara boards in a way that is compatible with DIOT.

However, Phaser and Fastino seem to be show-stopper here. I can see some workarounds, but they don't seem very convincing: 
- we could use dedicated gateware with Hi-Z state and after identification is done and OK, reboot to the final gateware,
- add MOSFET acting as analog LVDS switch so that they can be enabled along with P12V.

gkasprow, 2021-12-16:
We can also add inrush current protection to the adapter. Just resistor + bypass MOSFET. Simple and robust.

gkasprow, 2021-12-16:
I think the same anti-latch-up conditions must be added to the DIOT specification to avoid powering of boards by LVDS lines during startup. So some form of sequencing must be implemented anyway. Lack of proper sequencing could explain why Kasli FPGAs die from time to time...

gkasprow, 2021-12-16:
The correct sequencing could be implemented in the following way.
1. The controller has a power delivery circuit that wakes after time Tk. It's the time PG signal goes high.
2. The Peripheral boards must wake up later than Tk. It can be ensured by a simple MOSFET-based inrush current limiter. We specify Tp as a max possible time for peripheral boards
3. The controller enables LVDS lines after Tp

gkasprow, 2021-12-16:
In the case of many existing boards that consume little current, the Tk condition can be ensured by increasing the startup capacitor at the DC/DC converter. Some power-hungry ones may need an additional inrush circuit.

gkasprow, 2021-12-16:
@kaolpr 
> However, Phaser and Fastino seem to be show-stopper here. I can see some workarounds, but they don't seem very convincing:

I'm not that sure about them. They have FPGA with a complex PSU circuit. So they wake up slowly.
The real danger may lie in modules like DIO which wake up first, their drivers are strong (MLVDS) and they output 1.25V logic levels at all LVDS lines back-powering Kasli...

jordens, 2021-12-16:
That's another fundamental problem with the old EEM design. Let's fix it for cPCIS.
The Kasli LVDS transceivers can be turned off at runtime and should be until after the module is enabled.
Each module must remain off until power enabled by Kasli after optionally identifying it through I2C. I think we already support eeprom access without 12 V everywhere.

kaolpr, 2021-12-16:
I'm sort of skeptical about point 3 of the proposed sequence. 

Assuming we don't have dedicated separation buffers on controller  it imposes requirements for firmware. Either we make a generic "first-stage" firmware for power-up and identification with Hi-Z outputs and reboot to final one once process is finished or we must give a strict and complicated guidelines for developing firmware (so that outputs are Z on bootup and than start acting according to application).

If we can separate EEM from controller's FPGA in hardware (e.g. with MOSFET LVDS analog switch on adapter?) let's do it. It will save us a lot of firmware complexity.

gkasprow, 2021-12-16:
@kaolpr As @jordens wrote, LVDS in Kasli can be disabled at runtime. Adding MOSFETs to LVDS lines will worsen the signal integrity. And take a lot of space. And add significant cost.

kaolpr, 2021-12-16:
Well, after a though, it may not be so complex to implement. Fixed times seems OK in general.

kaolpr, 2021-12-21:
When it comes to power control from the controller, maybe we could also support `RST_N` signal? It can simply disable P12V

gkasprow, 2021-12-21:
this could potentially cause an issue when somebody toggles the supply of the peripheral device without disconnecting the LVDS drivers in FPGA...

kaolpr, 2021-12-23:
True. It must be kept disconnected.

gkasprow, 2021-12-23:
We can also make it in such a way that RST_N can only enable the P12V to help with sequencing...
But it's simpler to use geographic addressing and implement variable delays to make sure all peripheral boards are not powered at the same time. In this way, we can easily set the lowest delay for slot 1 and the highest one for slot 8. The step could be let's say 10ms