sinara-hw/DIOT_EEM_Adapter #9: Implement JTAG trick from DIOT FMC Carrier

kaolpr, 2021-12-16:
Maybe we could implement JTAG trick @marmeladapk implemented in FMC Carrier?

Based on `SERVMOD` state some of the LVDS lines are used to transport JTAG control to the controller module. It could be useful for Phaser / Urukul / Mirny as a remote update path.

marmeladapk, 2021-12-16:
I guess it should be implemented on each EEM as needed. Also, it's not DIOT specific, we can also use it in old EEM format. We just need a gpio expander with servmod line. I think it should be way faster than updating flash over I2C, as in phaser.

kaolpr, 2021-12-16:
I agree, however putting it as an assembly variant (we love variants, don't we :heart:) on the adapter we'd avoid creating new revisions for the time being. When all boards are updated we can remove it from the adapter.

BTW, really nice idea ;-).

marmeladapk, 2021-12-16:
Thanks!

But I don't get how we can implement it on an adapter. We would need a cable from adapter to JTAG connector.

kaolpr, 2021-12-16:
That's exactly what I was thinking about.

marmeladapk, 2022-12-19:
@mkiepiela Add reset signal to pin 14 as in digilent hs3.

mkiepiela, 2023-02-25:
TRSTn added to pin 14..

closed 2023-02-25 by mkiepiela