sinara-hw/DIOT_EEM_Adapter #10: Compatibility matrix

kaolpr, 2021-12-21:
|                   | DIOT Backplane | CPCIs Backplane | DIOT Peripheral | Sinara Peripheral | CPCIS peripheral | DIOT Controller | Kasli Controller | CPCIs Controller |
|:-----------------:|:--------------:|:---------------:|:---------------:|:-----------------:|:----------------:|:---------------:|:----------------:|:----------------:|
| DIOT Backplane    | :heavy_minus_sign:              |                 |                 |                   |                  |                 |                  |                  |
| CPCIs Backplane   | :heavy_minus_sign:              | :heavy_minus_sign:               |                 |                   |                  |                 |                  |                  |
| DIOT Peripheral   | :heavy_check_mark:              | :heavy_check_mark:               | :heavy_minus_sign:               |                   |                  |                 |                  |                  |
| Sinara Peripheral | :heavy_check_mark:              | :heavy_check_mark:               | :heavy_minus_sign:               | :heavy_minus_sign:                 |                  |                 |                  |                  |
| CPCIs Peripheral  | :grey_question:              | :heavy_check_mark:               | :heavy_minus_sign:               | :heavy_minus_sign:                 | :heavy_minus_sign:                |                 |                  |                  |
| DIOT Controller   | :heavy_check_mark:              | :heavy_check_mark:               | :heavy_check_mark:               | :heavy_check_mark:                 | :grey_question:                | :heavy_minus_sign:               |                  |                  |
| Kasli Controller  | :heavy_check_mark:              | :heavy_check_mark:               |  :grey_question:               | :heavy_check_mark:                 | :grey_question:                | :heavy_minus_sign:               | :heavy_minus_sign:                |                  |
| CPCIs Controller  | :grey_question:              | :heavy_check_mark:               | :x:               | :x:                 | :heavy_check_mark:                | :heavy_minus_sign:               | :heavy_minus_sign:                |                  |

Legend:

* :heavy_check_mark: - compatible configuration
* :heavy_minus_sign: - irrelevant configuration
* :grey_question: - may be incompatible (although may work in some cases, to some extent)
* :x: - incompatible configuration

kaolpr, 2021-12-21:
@gkasprow @marmeladapk Do you agree?

gkasprow, 2021-12-21:
CPCIS is not compatible with DIOT in any way because CPCIS is 160mm while DIOT is 220mm length.
One can order chassis from Schroff with a CPCIS backplane and 160 or 220 lengths.
we should not call it CPCIS any longer to not cause confusion.
We are using the CPCIS backplane only.
CPCIS is using PCI/USB/ETH interfaces, both from the controller and peripheral side.
We can only consider the compatibility between DIOT (6HP panels) and Sinara (4/8HP panels)
Schroff CPCIS-220mm has 4HP spacing, but we will use a much cheaper open-source CERN chassis and backplane.
6HP has an advantage that all existing boards can fit including Kasli, Stabilizer with mezzanine, Mirny with mezzanines.

kaolpr, 2021-12-23:
I've updated top comment to contain clear information on compatibility and removed redundant information from the table.

gkasprow, 2021-12-23:
LGTM
DIOT backplane is electrically 100% CPCIS backplane, The only difference is slot spacing.
The open-source [DIOT backplane ](https://ohwr.org/project/diot-crate-hw/wikis/home ) was demonstrated to work at 12Gbit/s.
I think we should event not consider 220mm  Schroff enclosures. They are expensive and proprietary which contradicts with Sinara ecosystem philosophy :) We can still use existing 4HP modules with adapters in DIOT. We simply use 2HP blind panels to separate the panels.

kaolpr, 2021-12-23:
Ok, so as you suggested:
>  we should not call it CPCIS any longer to not cause confusion.

kaolpr, 2021-12-23:
However, I'd keep the CPCIs incompatibility warning due to visual resemblance (module length can be overlooked).

gkasprow, 2021-12-23:
Let's rename the repository as well.

marmeladapk, 2022-12-19:
Added table from top post to wiki.

closed 2022-12-19 by marmeladapk