sinara-hw/DIOT_EEM_Adapter #14: DECT_BPR / DECT_RIO

kaolpr, 2021-12-21:
I'm not sure if we can connect `DECT_BPR` and `DECT_RIO` to GND. According to note (presumably quote from the CPCIs standard) it is backplane that connects those lines to ground, not peripheral.

[image]

Although in DIOT case that won't be a problem, as those lines are connected to GND on DIOT backplane, the same should not be expected from the CPCIs backplanes.

gkasprow, 2021-12-21:
what DIOT says about it?

kaolpr, 2021-12-22:
> those lines are connected to GND on DIOT backplane

Guess this feature is not used at all, as I have never heard of any other DIOT backplane.

gkasprow, 2021-12-23:
RIO is supported by DIOT. DIOT supports only single-star architecture. Some CPCIS backplanes also support full mesh, for Ethernet only. P6 is used for this purpose. The pins are NC at this moment.