sinara-hw/DIOT_EEM_Adapter #16: LVDS mux should be controlled by servmod

marmeladapk, 2022-11-30:
Controller should be able to switch LVDS lines to alternate mode. I'd like to keep this compliant with specification ([link](https://ohwr.org/project/diot/wikis/uploads/90afde7824b19862712b9903c5a68e9e/DIOT_system_specification_v2.0-dev-200626.pdf)) and consistent with other boards.

mkiepiela, 2022-12-12:
Controller is able to switch LVDS lines to alternate mode by setting IO expander controlled via I2C in Service Mode.
This solution was chosen together with @kaolpr as more flexible than switching muxes directly by SERVMOD line.
Notice that IO expander has to be controlled anyway (in Service Mode) when accessing the EEPROMs on adapter/EEM0/EEM1.

marmeladapk, 2022-12-12:
Yes, but this is not compatible with DIOT, so when mixing DIOT cards and cards with adapters you'll have two ways of controlling muxes. IMO it's better to stay consistent on this behaviour.

mkiepiela, 2022-12-15:
OK, I modified to keep compatibility with DIOT (switching muxes by SERVMOD as default behavior) but have some flexibility as well (possibility of switching muxes by IO expander).

closed 2022-12-19 by marmeladapk